energia/msp430-lg-core #4: Compiler warnings because of board naming.

robertinant, 2016-06-29:
_From @RickKimball on June 29, 2016 15:56_

The compiler produces a "command-line" warning because the '-' character is used in the board name
MSP-EXP430FR5969LP ( this is true for all boards with a dash)

$ "/mnt/vbox/shared/Downloads/energia-1.6.10E18B6/hardware/tools/msp430/bin/msp430-g++" -c -g -O2  -Wall -Wextra -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=msp430fr5969 -DF_CPU=8000000L -DARDUINO=10610 -DENERGIA=10610 -DENERGIA_MSP-EXP430FR5969LP -DENERGIA_ARCH_MSP430 -I/mnt/vbox/shared/Downloads/energia-1.6.10E18B6/hardware/tools/msp430/include  "-I/mnt/vbox/shared/Downloads/energia-1.6.10E18B6/hardware/energia/msp430/cores/msp430" "-I/mnt/vbox/shared/Downloads/energia-1.6.10E18B6/hardware/energia/msp430/variants/MSP-EXP430FR5969LP" "/tmp/build2eea903822402328ee22efe1219ad0fa.tmp/sketch/ASCIITable.ino.cpp" -o "/tmp/build2eea903822402328ee22efe1219ad0fa.tmp/sketch/ASCIITable.ino.cpp.o"
<command-line>:0:12: warning: missing whitespace after the macro name [enabled by default]

If you remove the '-' it compiles without warnings -DENERGIA_MSP_EXP430FR5969LP

Additionally, if you actually try to use the macro it creates more warnings
```
#ifdef ENERGIA_MSP-EXP430FR5969LP
 void some_function_conditionally_included(void)
 {
  return;
 }
 ##endif
```
/tmp/build1e32f1dce53658bb7281a24918dc9d1c.tmp/sketch/other.cpp:5:19: warning: extra tokens at end of #ifdef directive [enabled by default]

_Copied from original issue: robertinant/EnergiaNG#10_

robertinant, 2016-06-29:
Fixed with 73d5cbc520a0c1ca28edcceedecd83cc982f5f14

robertinant, 2016-06-29:
Will do the same for the other cores

rei-vilo, 2016-06-30:
Same as [Illegal Characters in Boards Tags #4](https://github.com/energia/tivac-core/issues/4) for Tiva C.

closed 2016-07-05 by robertinant

rei-vilo, 2016-12-31:
See http://forum.43oh.com/topic/10097-warning-missing-whitespace-after-the-macro-name/