energia/msp430-lg-core #7: Upload Using Programmer fails

RickKimball, 2016-07-03:
I tried using the Sketch/Upload Using Programmer menu item and it failed with this message:
```
...
Sketch uses 782 bytes (4%) of program storage space. Maximum is 16,384 bytes.
Global variables use 32 bytes (6%) of dynamic memory, leaving 480 bytes for local variables. Maximum is 512 bytes.
Error while uploading: missing 'program.tool' configuration parameter
```
OS: Linux Ubuntu 15.10
Using EnergiaNG build. ( I wasn't sure If I should file this under the IDE or the core )

robertinant, 2016-07-05:
Upload using programmer does not apply to Energia boards. Not sure how to handle this right now. I do want to keep this option in the menu for if users want to install cores other than Energia cores in the Energia IDE. I'll see if I can "fix" upload using programmer to do the same as just upload.

robertinant, 2016-07-05:
What does msp430-size report for that Sketch? I think the regex here https://github.com/energia/msp430-lg-core/blob/master/platform.txt#L80 is correct? Or am I missing something?

It could be that mspdebug is reporting other sections in the total size..

RickKimball, 2016-07-05:
Energia reports:
```
Sketch uses 782 bytes (4%) of program storage space. Maximum is 16,384 bytes.
Global variables use 32 bytes (6%) of dynamic memory, leaving 480 bytes for local variables. Maximum is 512 bytes.
```
msp430-size -A reports:
```
$ msp430-size -A Debounce.ino.elf 
Debounce.ino.elf  :
section           size    addr
.text              778   49152
.rodata             90   49930
.data                4     512
.bss                26     516
.noinit              2     542
.vectors            32   65504
.debug_aranges     336       0
.debug_info       7566       0
.debug_abbrev     1930       0
.debug_line       3548       0
.debug_frame       534       0
.debug_str        2647       0
.debug_loc        1598       0
.debug_ranges      140       0
Total            19231
```
The real size, which includes .text + .data + .rodata is this:
```
$ msp430-size Debounce.ino.elf 
   text	   data	    bss	    dec	    hex	filename
    900	      4	     28	    932	    3a4	Debounce.ino.elf
```

RickKimball, 2016-07-05:
BTW the flash size discrepancy is a different issue ( #8 )

robertinant, 2016-07-05:
Sorry posted in the wrong issue. Seems like .rodata is missing from:
recipe.size.regex=^(?:\.text|\.data|\.bootloader)\s+([0-9]+).*

I'll add it in the next release

RickKimball, 2016-07-05:
That only comes out to 872. You need to throw in the .vector sections too:

```
$ msp430-size -A Debounce.ino.elf | grep -e '.text' -e '.rodata' -e '.data' -e '.vectors' | awk  '{ sum += $2 }; END { print sum }'
904
$