energia/msp430-lg-core #8: Flash size reportedly used doesn't match actual size

RickKimball, 2016-07-03:
Using an msp430g2553 I created a small sketch.  The informational message says it is using 782 bytes however it is really using 872 bytes

```
Sketch uses 782 bytes (4%) of program storage space. Maximum is 16,384 bytes.
Global variables use 32 bytes (6%) of dynamic memory, leaving 480 bytes for local variables. Maximum is 512 bytes.
/mnt/vbox/shared/github/EnergiaNG/build/linux/work/hardware/tools/msp430/bin/mspdebug rf2500 --force-reset prog /tmp/build1e32f1dce53658bb7281a24918dc9d1c.tmp/Debounce.ino.hex 
MSPDebug version 0.22 - debugging tool for MSP430 MCUs
Copyright (C) 2009-2013 Daniel Beer <dlbeer@gmail.com>
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Trying to open interface 1 on 036
rf2500: warning: can't detach kernel driver: No data available
Initializing FET...
FET protocol version is 30066536
Set Vcc: 3000 mV
Configured for Spy-Bi-Wire
Sending reset...
Device ID: 0x2553
  Code start address: 0xc000
  Code size         : 16384 byte = 16 kb
  RAM  start address: 0x200
  RAM  end   address: 0x3ff
  RAM  size         : 512 byte = 0 kb
Device: MSP430G2553/G2403
Number of breakpoints: 2
fet: FET returned NAK
warning: device does not support power profiling
Chip ID data: 25 53
Erasing...
Programming...
Writing  872 bytes at c000...
Writing   32 bytes at ffe0...
Done, 904 bytes total

robertinant, 2016-07-05:
Fixed with ecb3bae3fee42e10e0467d8ecf112747de15ccba

RickKimball, 2016-07-05:
see comments in issue #7

robertinant, 2016-07-18:
I have added the .vectors section as well.

closed 2016-07-18 by robertinant