energia/msp430-lg-core #12: usbutil: unable to find a device matching 0451:f432

robertinant, 2016-08-22:
See details here: http://forum.43oh.com/topic/9848-energia-1610r18-windows-7-launchpad-msp430g2553-dont-load-sketch/?p=74156

robertinant, 2017-08-08:
Fixed. Resolution > Updated mspdebug upload settings and modified mspdebug to always use tilib on windows even though rf2500 is specified as driver.

closed 2017-08-08 by robertinant

stilren, 2017-08-10:
@robertinant  Where can we get the release (where this is fixed? Energia 1.6.10E18 is still the version on the download page (http://energia.nu/download/)

davemorl, 2017-09-28:
What a disgrace this is from energia and texas instruments. The last release was in November last year and this is still not fixed. I'm tempted to just throw my launchpad in the bin!!!

robertinant, 2017-09-28:
This has been fixed. Not all fixes have to go trhough an updated IDE. Most fixes happen through a core update including this isse. Please update the msp430 core through the Board Manger: Tools->Board Manager. In the board manager dialog look for MSP430. Select it and the click update. A new core and it's dependencies will be installed. After the update you will find that this issue has been fixed.

FabianaTrigo, 2017-11-24:
@robertinant Tks for help!!!

instru003, 2018-03-17:
Thank you robertinant. It worked.

avivalbeg, 2018-03-24:
robertinant, How do you update the upload settings?