energia/msp430-lg-core #15: MSP430F5529 : undefined reference to `operator new[](unsigned int)'

rei-vilo, 2016-08-25:
Using `new` against the MSP430F5529 raises an error.

```
undefined reference to `operator new[](unsigned int)'
```

Maybe [energia/msp430-ng-core](https://github.com/energia/msp430-ng-core) will support it?

Cross posted with

robertinant, 2016-08-25:
Can you please provide an example Sketch that shows the error?

rei-vilo, 2016-08-25:
```
uint8_t * pointer;

void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600);

  uint8_t * table = new uint8_t[16];
  pointer = (uint8_t *) table;

  pointer [1] = 1;
}

void loop() {
  // put your main code here, to run repeatedly:
  Serial.print(pointer[1], DEC);
  while (1);
}
```

robertinant, 2016-08-29:
mspgcc does not have full support for C++. For example, there is no libstdc++. The new compiler soon to be featured in Energia will have more extensive support for C++.

rei-vilo, 2016-12-05:
For the moment, the only turn-around is to define the arrays directly as variable without using `new()`.

rei-vilo, 2017-01-08:
@robertinant Any update on the switch to the new compiler?

RickKimball, 2017-04-03:
fixed with: https://github.com/energia/msp430-lg-core/commit/9b6fc95750b87eed164907902bfa23be29c01576

closed 2017-04-03 by RickKimball