energia/msp430-lg-core #18: Default Analog Read Resolution Inconsistent

robertinant, 2016-08-29:
_From @croberts15 on January 28, 2016 17:1_

The default analog read resolution should be consistent across all boards at 10 bit, however when testing I was getting a default of 12-bit for 5529, 6989, and tm4c123.  The msp432 was 10-bit. 

_Copied from original issue: energia/Energia#834_

robertinant, 2016-08-29:
_From @rei-vilo on January 28, 2016 17:14_

See #822 for CC3200.

robertinant, 2016-11-07:
This is a potential dangerous change. If I change it then users who upgrade will now suddenly get 10 bit vs 12 bit. We can document this but I many users might not read the documentation. Thoughts?

rei-vilo, 2016-11-07:
Add a function that returns the number of bits for ADC. So the user knows which is the maximum.