energia/msp430-lg-core #26: Have a way to avoid enableXtal

barawn, 2016-10-27:
enableXtal is a pretty costly function - it runs at startup and if the crystal isn't present, it delays code execution for 2 seconds. It'd be really nice if there were a way to avoid it.

In my recent projects, what I've done is expose "enableXtal" in Energia.h, and give it a weak attribute in wiring.c. This allows me to override it in the main sketch with an empty function, which gets rid of the 2 second delay at startup, but examples that do depend on the crystal work perfectly fine.

RickKimball, 2017-04-03:
I like this solution, any objection to this approach @robertinant ?

robertinant, 2017-04-03:
I do to like this too!. No objections at all. This way advanced users will be able to control the clock sources as they see fit whereas novice users will still get the "auto" clock selection for low power modes and don't necessarily care about the 2 second delay at startup if the crystal is not present. Let's go for it and then document on energia.nu.

barawn, 2017-04-03:
OK, I just submitted a pull request #39  with these changes.

robertinant, 2017-08-08:
Pull request was merged. Closing issue.

closed 2017-08-08 by robertinant