energia/msp430-lg-core #28: FR5994 — Two Analog Inputs Share the Same Pin

rei-vilo, 2016-11-16:
In file [MSP-EXP430FR5994LP/pins_energia.h](https://github.com/energia/msp430-lg-core/blob/FR5994_FR2311_support/variants/MSP-EXP430FR5994LP/pins_energia.h) lines [106](https://github.com/energia/msp430-lg-core/blob/FR5994_FR2311_support/variants/MSP-EXP430FR5994LP/pins_energia.h#L106) and [110](https://github.com/energia/msp430-lg-core/blob/FR5994_FR2311_support/variants/MSP-EXP430FR5994LP/pins_energia.h#L110), two analog inputs `A5` and `A9` share the same pin `28`.

```
static const uint8_t A5  = 28;
static const uint8_t A9  = 28;
```

rei-vilo, 2016-11-16:
As per [SLAU678a](http://www.ti.com/lit/ug/slau678a/slau678a.pdf), shouldn't be `P4_1 = `A9`= pin `31`?

StefanSch, 2016-11-16:
fixed with last commit

closed 2016-11-16 by StefanSch