energia/msp430-lg-core #31: How to use FRAM as RAM with Energia?

rei-vilo, 2016-12-09:
FRAM is an exclusive technology, but how to leverage it for large arrays?

Please document how to use FRAM as RAM with Energia.

spirilis, 2016-12-09:
I had a thread on 43oh for Wolverine FRAM tips & tricks that might still apply...

rei-vilo, 2016-12-09:
@spirilis Thank you for the pointer.

I've already saw and used the [trick](http://forum.43oh.com/topic/5474-energia-and-wolverine-tips/?p=47861) but I want it included and fully supported in Energia.

```
#define PERSIST __attribute__((section(".text")))
uint8_t DisplayBuffer[LCD_MAXIMUM_Y][LCD_MAXIMUM_X] PERSIST;
```

spirilis, 2016-12-09:
Gotcha... Guess that would just mean Energia including the #define in its Energia.h file?

rei-vilo, 2016-12-09:
Yes, plus some doc!

rei-vilo, 2017-01-08:
Add `FRAM` attribute to Energia in the `pins_energia.h` related to FRAM-based LaunchPads.

```
#define FRAM __attribute__((section(".text")))
```

rei-vilo, 2017-01-08:
See pull-request #32

StefanSch, 2017-01-26:
I am a little bit burned child with short names esp. in defines and that they got reused.
May we can find a better / unique name, e.g.
STORE_IN_FRAM
PLACE_IN_FRAM
ALLOCATE_IN_FRAM
IN_FRAM
...

rei-vilo, 2017-01-26:
`PLACE_IN_FRAM`seems fine, unless @robertinant has a better suggestion.

rei-vilo, 2017-01-26:
See https://github.com/energia/msp430-lg-core/pull/32#issuecomment-275451946

rei-vilo, 2017-11-20:
The FRAM is write protected on the MSP430FR2433. See #47.

closed 2017-11-20 by rei-vilo