energia/msp430-lg-core #34: MSP430G2553 — Flash +19% and RAM +57%

rei-vilo, 2017-01-30:
The same `blink` sketch compiled executable inflates 19%, and RAM usage increases +57%. 

This is especially critical for the  **MSP430G2553** with 512 bytes available!

* Using `msp430/bin/msp430-size`, 

```
   text	   data	    bss	    dec	    hex	filename
    758	      0	     14	    772	    304	Blink3.ino.elf with Energia 12
```

```
   text	   data	    bss	    dec	    hex	filename
    900	      0	     22	    922	    39a	Blink3.ino.elf with Energia 18 and MSP430 1.0.2
```

* Code

```
uint8_t myLED;

void blink(uint8_t pin, uint8_t times, uint16_t ms, bool level = true);

void setup()
{
    myLED = RED_LED;
    pinMode(myLED, OUTPUT);
}

void loop()
{
    blink(myLED, 3, 333);
    delay(1000);
}

void blink(uint8_t pin, uint8_t times, uint16_t ms, bool level)
{
    for (uint8_t i = 0; i < times; i++)
    {
        digitalWrite(pin, level ? HIGH : LOW);
        delay(ms >> 1);
        digitalWrite(pin, level ? LOW : HIGH);
        delay(ms >> 1);
    }
}
```

robertinant, 2017-01-30:
Did you use the old or new compiler for this?

rei-vilo, 2017-01-30:
I used the old compiler part of `msp430-lg-core` and went through releases 12 to 18 of Energia to check the size. 

> msp430-gcc (MSPGCC 20120406 (With patches: sf3540953 sf3559978)) 4.6.3 20120301 (mspgcc LTS 20120406 unpatched)

robertinant, 2017-01-30:
platform.txt has it's optimization set to -O2. What sizes do you get if you change this to -Os for compiler.c.flags and compiler.cpp.flags

rei-vilo, 2017-01-30:
I'm using embedXcode and both are set with `-0s`.

The issue isn't critical per se, as embedXcode gives me the flexibility to use the previous MSP430 package of Energia 12.

Everything is identical, except the release of the framework, `0.1.2` or `1.0.2`.

* With Energia 12

```
/Applications/IDE/Energia.app/Contents/Java/hardware/tools/msp430/bin/msp430-g++ -c -Os -w -c -Os -Wall -ffunction-sections -fdata-sections -mmcu=msp430g2553 -DF_CPU=16000000L -I/Users/ReiVilo/Library/Energia15/packages/energia/hardware/msp430/0.1.2/variants/launchpad -I/Users/ReiVilo/Library/Energia15/packages/energia/hardware/msp430/0.1.2/cores/msp430 -I/Users/ReiVilo/Library/Energia15/packages/energia/hardware/msp430/0.1.2/cores/msp430/avr -I/Users/ReiVilo/Library/Energia15/packages/energia/hardware/msp430/0.1.2/libraries -I/Applications/IDE/Energia.app/Contents/Java/libraries -DENERGIA=10610 -DARDUINO=10610 -DEMBEDXCODE=614 -DENERGIA_ARCH_MSP430 -DENERGIA_lpmsp430g2553 -D__MSP430G2553__     -fno-exceptions -fno-threadsafe-statics  /Users/ReiVilo/Library/Energia15/packages/energia/hardware/msp430/0.1.2/cores/msp430/HardwareSerial.cpp -o Builds/cores/msp430/HardwareSerial.cpp.o
```

* With Energia 18 and MSP430 package release 1.0.2

```
/Applications/IDE/Energia.app/Contents/Java/hardware/tools/msp430/bin/msp430-g++ -w -c -Os -Wall -ffunction-sections -fdata-sections -mmcu=msp430g2553 -DF_CPU=16000000L -I/Users/ReiVilo/Library/Energia15/packages/energia/hardware/msp430/1.0.2/variants/MSP-EXP430G2 -I/Users/ReiVilo/Library/Energia15/packages/energia/hardware/msp430/1.0.2/cores/msp430 -I/Users/ReiVilo/Library/Energia15/packages/energia/hardware/msp430/1.0.2/cores/msp430/avr -I/Users/ReiVilo/Library/Energia15/packages/energia/hardware/msp430/1.0.2/libraries -I/Applications/IDE/Energia.app/Contents/Java/libraries -DENERGIA=10610 -DARDUINO=10610 -DEMBEDXCODE=614 -DENERGIA_ARCH_MSP430 -DENERGIA_MSP-EXP430G2553LP -D__MSP430G2553__     -fno-exceptions -fno-threadsafe-statics  /Users/ReiVilo/Library/Energia15/packages/energia/hardware/msp430/1.0.2/cores/msp430/HardwareSerial.cpp -o Builds/cores/msp430/HardwareSerial.cpp.o
```

robertinant, 2017-01-30:
What numbers do you get with 1.0.1?

rei-vilo, 2017-01-30:
Energia |    text |    data |     bss |     dec |     hex
--------|--------|--------|--------|--------|--------
12 | 758 | 0 | 14 | 772 | 304
13 | 896 | 0 | 22 | 918 | 396
14 | 896 | 0 | 22 | 918 | 396
15 | 896 | 0 | 22 | 918 | 396
16 | 896 | 0 | 22 | 918 | 396
17 | 900 | 0 | 22 | 922 |     39a
18 / 1.0.1 | 900 | 0 | 22 | 922 |     39a
18 / 1.0.2 | 900 | 0 | 22 | 922 |     39a

RickKimball, 2017-01-30:
Check out the -minrt flag combined with -Os

http://forum.43oh.com/topic/5767-alternative-super-small-crt-for-msp430-elf-gcc-491/?p=50352

rei-vilo, 2017-01-30:
Results unchanged, 758 / 14 for Energia 12, 900 / 22 after.

Flash size isn't a big deal, question is about RAM. With 512 bytes, each byte counts!

Some projects fail to run on the MSP432G2553, unless I use Energia 12.

robertinant, 2017-01-30:
This is only going to get worse with the new gcc compiler I am afraid..

rei-vilo, 2017-01-30:
No problem, as long as I can keep and use the old one with embedXcode.

robertinant, 2017-01-30:
Yes, we are going to support both the new and the old compiler with the same core.

robertinant, 2017-08-08:
This was due to the -g flag being supplied and has been fixed in the upcoming release later this week.

```
Sketch uses 810 bytes (4%) of program storage space. Maximum is 16,384 bytes.
Global variables use 20 bytes (3%) of dynamic memory, leaving 492 bytes for local variables. Maximum is 512 bytes.
```

closed 2017-08-08 by robertinant