energia/msp430-lg-core #36: Supply board manager JSON files

barawn, 2017-02-28:
It would be good if the MSP430 package indexes were available for the boards that get pre-installed in Energia, so that we can have a working example to start from. I've been totally unable to make a custom board install via the Board Manager since I'm pretty much workinng blind.

rei-vilo, 2017-02-28:
I'm not sure to understand your question. The list of the pre-installed MSP430 boards is available in 4 different locations:

* Boards Manager window under menu **Tools > Board > Boards Manager** 

[image]

* Boards list under menu **Tools > Board**

[image]

* File `boards.txt`under folder `~/Library/Energia15/packages/energia/hardware/msp430/1.0.2/boards.txt`

* File `boards.txt`folder
`/Applications/IDE/Energia.app/Contents/Java/hardware/energia/msp430/boards.txt` folder

closed 2017-02-28 by rei-vilo

barawn, 2017-03-01:
I was asking for the package_index.json file so that it can be used as a model for building board manager packages. Since I don't know what the tools are called (in Board Manager-terms) that are installed, I can't make one of my own.

I did find it, though, buried in a temp directory (on Windows at least), but it would probably be a good idea to expose that package index somewhere where people can see it.

rei-vilo, 2017-03-01:
Energia uses the same structure as Arduino, so all the Arduino documentation is relevant. 

See for example Boards Manager JSON index file format specification at
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.6.x-package_index.json-format-specification

barawn, 2017-03-01:
Right, but without that JSON file, you don't know what tools are already included in Energia, or where they're stored from. In the JSON on the temp directory, for instance, which I have no idea where it came from, it specifies the toolsDependencies as  "name":"msp430-gcc", "version":"4.6.3", "name":"dslite", "version":"6.2.1.1594", etc. So when I create one for my custom boards, I can use exactly those toolsDependencies so they're automatically satisfied by what's already installed.

rei-vilo, 2017-03-01:
It is a practice to release self-contained packages, as Arduino does for the AVR package, stored along other packages instead of within the IDE.

robertinant, 2017-03-01:
This is the repo for package_index.json https://github.com/robertinant/energia-package-index
The json file is downloaded by the IDE from http://energia.nu/packages/package_index.json. You can specify your own custom board packages index in the IDE's preferences. You will need to name it something other than package_index.json e.g. my_package_index.json.

barawn, 2017-03-01:
Fantastic! Thanks, I didn't see that repository before.