energia/msp430-lg-core #37: macOS: MSP430F5529 Debug Interface Version

rei-vilo, 2017-03-24:
On macOS, 

* Energia 1.6.1E0018 with `msp430-lg-core 1.0.1` boards package, and 
* CCS Cloud 

no longer share the same release of the debug interface for the **MSP430F5529**.

[image]

[image]

robertinant, 2017-03-24:
I am working on getting them back in sync.

robertinant, 2017-08-08:
An updated msp430 package will be released later this week which get's them back in sync. I'll do my best to keep them in sync.

closed 2017-08-08 by robertinant