energia/msp430-lg-core #46: Error compiling for board MSP-EXP430FR5994LP with release 1.0.2

rei-vilo, 2017-08-25:
Error compiling for board MSP-EXP430FR5994LP.

>/Applications/IDE/Energia.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware "/Applications/IDE/Energia.app/Contents/Java/hardware" -hardware "/Users/ReiVilo/Library/Energia15/packages" -tools "/Applications/IDE/Energia.app/Contents/Java/tools-builder" -tools "/Applications/IDE/Energia.app/Contents/Java/hardware/tools/msp430" -tools "/Users/ReiVilo/Library/Energia15/packages" -built-in-libraries "/Applications/IDE/Energia.app/Contents/Java/libraries" -libraries "/Users/ReiVilo/Documents/Projets/Energia/libraries" -fqbn=energia:msp430:MSP-EXP430FR5994LP -ide-version=10610 -build-path "/var/folders/px/cyfvtr757lqg0yp_cv9j79jh0000gn/T/build2b41f88c4b9135c49eb220355a1bdfa8.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "/var/folders/px/cyfvtr757lqg0yp_cv9j79jh0000gn/T/untitled1017913555.tmp/sketch_aug25a/sketch_aug25a.ino"
>
>Board MSP-EXP430FR5994LP (platform msp430, package energia) is unknown
>
>Error compiling for board MSP-EXP430FR5994LP.
>

robertinant, 2017-08-25:
I am not able to reproduce this. How did you install the core? I just did a clean install of the core and it compiles OK for all boards.

rei-vilo, 2017-08-25:
I perform a new clean install and the problem was solved.

closed 2017-08-25 by rei-vilo

robertinant, 2017-08-25:
hmm. How did you clean the install? I just now tried on another Mac which is 100% clean and was not able to reproduce it either.

rei-vilo, 2017-08-25:
I went back to release 1.0.1, which is not located under the packages but inside Energia.app, and then asked for an update.

By the way, the FR2311 isn't activated.

robertinant, 2017-08-25:
FR2311 has to little memory to be useful in Energia. You can however comment it out in boards.txt to activate it.