energia/msp430-lg-core #52: MSP430FR5739 Conflicting Pins Definitions

rei-vilo, 2018-01-22:
Lines 84-86 and 92-94 are conflicting on [pins_energia.h](https://github.com/energia/msp430-lg-core/blob/0bb691fce4bddcc47c67e6e951cefe6d4ae955e6/variants/MSP-EXP430FR5739LP/pins_energia.h) for MSP430FR5739.

```
static const uint8_t A4  = 18; /* 18 - P3.2 */
static const uint8_t A5  = 17; /* 17 - P3.1 */
static const uint8_t A6  = 16; /* 16 - P3.0 */
```

and

```
static const uint8_t ACC_X  = 12; /* (A6) 16 - P3.0 */
static const uint8_t ACC_Y  = 13; /* (A5) 17 - P3.1 */
static const uint8_t ACC_Z  = 14; /* (A4) 18 - P3.0 */
```

Using `AnalogRead(ACC_X)` doesn't return a valid value, `AnalogRead(16)` does.

robertinant, 2018-01-22:
It should be: 
```
static const uint8_t ACC_X  = 16; /* (A6) 16 - P3.0 */
static const uint8_t ACC_Y  = 17; /* (A5) 17 - P3.1 */
static const uint8_t ACC_Z  = 18; /* (A4) 18 - P3.0 */
```
The naming of the analog channels on the headers does not follow the name of the actual MSP430  analog channels. e.g. P3.0 is A12 for the MSP430 but is named A6 for the LaunchPad headers. Not sure what the motivation was 4 years ago but unfortunately it can not be changed. What can be changed is the ACC_X, ACC_Y and ACC_Z. If agreed on the above I will make the change and push it.

rei-vilo, 2018-01-22:
Great, see pull-request https://github.com/energia/msp430-lg-core/pull/53.

closed 2018-01-22 by rei-vilo