energia/msp430-lg-core #54: MSP430FR4311 — Issue with Serial

rei-vilo, 2018-01-31:
The following basic sketch doesn't display anything on none of the two serial ports from the MSP430FR4311 LaunchPad. The same sketch works fine on the MSP430FR5969 LaunchPad. All jumpers except CTS and RTS are set. I 'm facing the same issue on both macOS and Windows.

```
#define LED RED_LED

void setup() {                
  pinMode(LED, OUTPUT); 
  Serial.begin(9600);    
}

void loop() {
  Serial.println(millis());
  digitalWrite(LED, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);               // wait for a second
  digitalWrite(LED, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);               // wait for a second
}
```

StefanSch, 2018-01-31:
Unfortunately the RED LED and the UART TX are mapped to the same pin.
So each time you toggle the RED LED the UART pin is reconfigured.
Use #define LED GREEN_LED

closed 2018-01-31 by StefanSch

rei-vilo, 2018-01-31:
Thanks! I've just forget about this confusing RED LED and UART TX mapped to the same pin!