energia/msp430-lg-core #55: MSP430FR4311 — Issue with I²C

rei-vilo, 2018-01-31:
The MSP430FR4311 uses software I²C on pins 9 and 10 for compatibility. However, only SCL signal seems to be generated.

The same sketch with the same INA226 BoosterPack works fine on the MSP430FR6989 LaunchPad.

I'm using MSP430 boards package release 1.0.3.

[image]

```
#include "Wire.h"

#define INA226_I2C_ADDRESS 0x40
#define INA226_REGISTER_MANUFACTURER_ID 0xfe
#define INA226_REGISTER_DIE_ID          0xff

int16_t read16(uint8_t address)
{
  int16_t value = 0;

  Wire.beginTransmission(INA226_I2C_ADDRESS);
  Wire.write(address);
  Wire.endTransmission();

  Wire.requestFrom(INA226_I2C_ADDRESS, 2);
  while (Wire.available() < 2)
  {
    Serial.print(".");
  }

  value = Wire.read();
  value <<= 8;
  value += Wire.read();

  return value;
}

void setup() {
  Wire.begin();
  Serial.begin(9600);
}

void loop() {
  Serial.print("5449 ? ");
  Serial.println(read16(INA226_REGISTER_MANUFACTURER_ID), HEX); // 5449
  Serial.print("2260 ? ");
  Serial.println(read16(INA226_REGISTER_DIE_ID), HEX); // 2260
  delay(100);

}
```

StefanSch, 2018-01-31:
Can you please check if both (SDA and SCL) have pull up resistor on the BoosterPack.

rei-vilo, 2018-01-31:
Yes, the [schematics](https://e2e.ti.com/cfs-file/__key/communityserver-discussions-components-files/968/11_2D00_-WCS033A_2800_Boosterpack_29005F00_Sch.PDF) shows both SCL and SDA have 10 kΩ pull-up resistors.

[image]

rei-vilo, 2018-01-31:
Same happens with the Sensors BoosterPack and the `WeatherSensors_Demo` example.

> TMP007_internalTemperature 0.00°C
> TMP007_externalTemperature 0.00°C
> OPT3001_light             0.00lux
> BME280_pressure           0.00hPa
> BME280_temperature        0.00°C
> BME280_humidity           0.00%

[image]

StefanSch, 2018-02-02:
The sensor BoosterPack has the I2C on pins 14 and 15 while the default for Energia is 9 and 10.

So you have to add
Wire.setModule(0);

StefanSch, 2018-02-02:
Hope this helps - if not let me know and i will debug into it.

rei-vilo, 2018-02-02:
Here are the results of my tests

- Test MSP430FR2433 I²C with INA226

Wire.setModule | Pins | Result
---- | ---- | ----
Wire.setModule(0); | 14 + 15 | 5449 =? 5449
Wire.setModule(1); | 14 + 15 | 5449 =? 0
Wire.setModule(0); | 9 + 10 | 5449 =? ................ERROR
Wire.setModule(1); | 9 + 10 | 5449 =? 0

- Test MSP430FR6989 I²C with INA226

Wire.setModule | Pins | Result
---- | ---- | ----
Wire.setModule(1); | 9 + 10 | 5449 =? 5449
No Wire.setModule = default = 1 | 9 + 10 | 5449 =? 5449

- Code

```
#include "Wire.h"

#define INA226_I2C_ADDRESS 0x40
#define INA226_REGISTER_MANUFACTURER_ID 0xfe

void setup()
{
    Serial.begin(9600);
    Wire.setModule(0);
    Wire.begin();
    delay(100);

    Serial.println("Test MSP430FR2433 I2C with INA226");
    Serial.println("Wire.setModule(0);");
    Serial.println("----");
    Serial.print("5449 =? ");

    Wire.beginTransmission(INA226_I2C_ADDRESS);
    Wire.write(INA226_REGISTER_MANUFACTURER_ID);
    Wire.endTransmission();

    uint16_t count = 16;
    uint16_t value = 0;
    Wire.requestFrom(INA226_I2C_ADDRESS, 2);
    while ((Wire.available() < 2) and (count > 0))
    {
        delay(100);
        Serial.print(".");
        count--;
    }

    if (count > 0) {
        value = Wire.read();
        value <<= 8;
        value += Wire.read();
        Serial.println(value, HEX);
    } else {
        Serial.println("ERROR");
    }
    Serial.println("====");

}

void loop()
{

}
```

StefanSch, 2018-02-04:
A little bit confused now - which launchpad shows the error: 
MSP430FR4311 (title say 4133 but that does not exist)
MSP430FR2433 (as in the last comment)?

rei-vilo, 2018-02-04:
Sorry, the LaunchPad with issues is the MSP430FR4133. 
The reference LaunchPad that works is the MSP430FR6989.

StefanSch, 2018-02-05:
fixed in branch issue_55

closed 2018-02-05 by StefanSch