energia/msp430-lg-core #61: How to Add Support for MSP430G2955

rei-vilo, 2018-04-02:
How difficult / easy would it be to add support for the MSP430G2955 on Energia?

Apart the MSP430G2553 which comes in DIP package, the MSP430G2955 is easy to solder on a TSSOP to DIP adapter. It provides more Flash and RAM.

[image]

Thank you!

StefanSch, 2018-04-03:
Adding support for the G2955 would be quite easy, the question is more the pin map which should be used as Energia is centered around a Board/Launchpad on the pin map.

rei-vilo, 2018-04-03:
We could stick with the pin numbering scheme of the MCU, as there is no LaunchPad available.

[image]

StefanSch, 2018-04-24:
just pushed a initial version to the G2955_support branch

rei-vilo, 2018-04-24:
Thank you very much, @StefanSch. I'll a try and let you know!

robertinant, 2018-04-24:
Thanks @StefanSch for adding support for the G2955. One question @rei-vilo. Should we put this in the boards menu for all or hide en you will in unhide it for your purposes?
Either works for me.

rei-vilo, 2018-04-26:
So far:

+ Blink LED works on `P1_0` = `RED_LED`
+ Serial works
+ SPI to Nokia 5110 screen works

_To be continued..._

rei-vilo, 2018-04-26:
## G2955 connected to Nokia 5110 screen through SPI

[image]

rei-vilo, 2018-05-31:
All the tests conducted on my side are positive, including UART, SPI and I²C as per [Standalone MSP430G2955](https://embeddedcomputing.weebly.com/standalone-msp430g2955.html).

[image]

@robertinant Wouldn't be nice to add the MSP430G2955 to the main distribution?