energia/msp430-lg-core #62: Hardware I²C No Longer Works on MSP432G2553

rei-vilo, 2018-05-18:
Targeting the MSP430G2553 and using the BME280 sensor.

- Emulated I²C on pins 9 + 10 works with Energia 17.

- Hardware I²C on pins 14 + 15 does not work with Energia 17.

- Hardware I²C on pins 14 + 15 works with Energia 12.

rei-vilo, 2018-08-05:
Same issue on the MSP430FR5994.

StefanSch, 2018-08-07:
There is a fix already commited into the master branch but not in the release.
Using this i do not see an issue on FR5994 I2C on pin 14/15 (did not test 9/10)

closed 2018-11-15 by StefanSch

rei-vilo, 2018-12-04:
No, it is just that legacy I²C (hardware, master+ slave) no longer worked on pins 14 + 15.

StefanSch, 2018-12-05:
All device now have I2C default on 9/10. But most of them can selected another module (SetModule) and then also use the legacy I2C on 14/15.
If 9/10 do not have hardware I2C pins mapped there it will use the SW functions.