energia/msp430-lg-core #63: Support for GCC MSP430 6.4.0.32?

rei-vilo, 2018-05-31:
Current GCC tool-chain for MSP430 is versioned `4.6.5` but is limited to 64 KB. 

I gave a try to GCC MSP430 `6.4.0.32` with options `-mlarge -mcode-region=upper -mhwmult=auto`.

Surprise is, variable and code sizes are significantly bigger.

* With GCC MSP430 `4.6.5` for _blink_ against MSP430G2553

> Estimated Flash:  1114 bytes used (2.3% of 48128 maximum), 47014 bytes free (97.7%)
> Estimated SRAM:   22 bytes used (1.0% of 2048 maximum), 2026 bytes free (99.0%)
> 

* With GCC MSP430 `6.4.0.32` for _blink_ against MSP430G2553

> Estimated Flash:  1936 bytes used (4.0% of 48128 maximum), 46192 bytes free (96.0%)
> Estimated SRAM:   152 bytes used (7.4% of 2048 maximum), 1896 bytes free (92.6%)
> 

This may rise issues for the MSP430G2553 but not for the MSP430F5529, MSP430FR5969, MSP430FR5994 and MSP430FR6989 with large Flash and RAM.

I haven't explored yet project which size is larger than 64 KB.

Are there a roadmap and a release date for the support of GCC MSP430 `6.4.0.32` with projects larger than 64 KB?

Thank you!

rei-vilo, 2018-06-15:
Latest release 8.1.0 of Code Composer Studio includes MSP430 GCC v7.3.1.24.

See **New in CCS v8.1.0** at [Release Notes for Code Composer Studio™ v8.1.0](http://software-dl.ti.com/ccs/esd/CCSv8/CCS_8_1_0/exports/CCS_8.1.0_ReleaseNotes.htm).

rei-vilo, 2018-10-09:
@StefanSch Does GCC 4.6.6 offer access to the whole memory beyond 48 KB? 

The idea is to get access to the full size of the FRAM memory, up to 256 KB with the MSP430FR5994.

StefanSch, 2018-10-10:
Unfortunately the currently used gcc compiler does not support the upper memory.
To support this the change to a later compiler version would be required which is still pending.
Most important point for not having switched to a newer compiler version is the increased code size the new compiler will bring and therefore limit functionality on some of the small memory devices.