energia/msp430-lg-core #68: Add Support for Second UART

rei-vilo, 2018-08-27:
For example, only one UART `eUSCI_A0 UART` on LaunchPad pins `3`=`RX` and `4`=`TX` is implemented on the **MSP430FR2433**.

How to use the second hardware UART `eUSCI_A1 UART` on LaunchPad pins `14`=`RX` and `15`=`TX`?

Thank you!

dberenguer, 2018-08-27:
For MSP430FR2433 the following definitions need to be added in pins_energia.h:

`static const uint8_t AUX_UARTRXD = 14;  /* Receive  Data (RXD) at P2.5 */
static const uint8_t AUX_UARTTXD = 15;  /* Transmit Data (TXD) at P2.6 */
#define AUX_UARTRXD_SET_MODE (PORT_SELECTION0 | INPUT)
#define AUX_UARTTXD_SET_MODE (PORT_SELECTION0 | OUTPUT)
#define AUX_UART_MODULE_OFFSET 0x20
#define SERIAL1_AVAILABLE 1`

Apart from clarifying that Serial1 and SPI can not be used at the same time.

rei-vilo, 2018-08-27:
@dberenguer ¡Gracias Daniel!

StefanSch, 2018-09-10:
thanks - added this also into the repository

closed 2018-09-10 by StefanSch