energia/msp430-lg-core #70: PLACE_IN_FRAM Error Message

rei-vilo, 2018-10-03:
Using `PLACE_IN_FRAM` against the MSP430FR5994 LaunchPad throws this error message:

```
/var/folders/px/cyfvtr757lqg0yp_cv9j79jh0000gn/T//ccaCFwLd.s: Assembler messages:
/var/folders/px/cyfvtr757lqg0yp_cv9j79jh0000gn/T//ccEVAsRJ.s:218: Warning: ignoring changed section attributes for .text
```

I'm using the latest release of [msp430-lg-core/variants/MSP-EXP430FR5994LP/pins_energia.h](https://github.com/energia/msp430-lg-core/blob/master/variants/MSP-EXP430FR5994LP/pins_energia.h#L42)

https://github.com/energia/msp430-lg-core/blob/0a245b29fc047d208982f383be1811a3ea1b25d8/variants/MSP-EXP430FR5994LP/pins_energia.h#L42

How to understand the error message and how to fix it?

See #49 and #31.

rei-vilo, 2018-10-04:
Same with CCS.

The code

```
uint8_t frameNew[4096] __attribute__((section(".text")));
```

returns the message

```
Description	Resource	Path	Location	Type
ignoring changed section attributes for .text	.ccsproject	/FRAM_EPD_Controller	line 321, external location: C:\Users\ReiVilo\AppData\Local\Temp\ccC7kneT.s	C/C++ Problem
```

It is not clear whether the message is a warning or an error.

StefanSch, 2018-10-08:
Just made a quick test with below code but could not see an issue.
Can you check with this again?

// most launchpads have a red LED
#define LED RED_LED

int i PLACE_IN_FRAM ;

// the setup routine runs once when you press reset:
void setup() {                
  // initialize the digital pin as an output.
  pinMode(LED, OUTPUT);     
  i = 0;
}

// the loop routine runs over and over again forever:
void loop() {
  if (i++ == 10)
  { 
    i=0;
  
    digitalWrite(LED, HIGH);   // turn the LED on (HIGH is the voltage level)
    delay(1000);               // wait for a second
    digitalWrite(LED, LOW);    // turn the LED off by making the voltage LOW
    delay(1000);               // wait for a second
  }
}

rei-vilo, 2018-10-08:
The above sketch still outputs the warning / error

> C:\Users\ReiVilo\AppData\Local\Temp\cc6j4iC7.s: Assembler messages:
> C:\Users\ReiVilo\AppData\Local\Temp\cc6j4iC7.s:57: Warning: ignoring changed section attributes for .text

Configuration
+ Energia 1.6.8E18 
+ MSP430 boards package 1.0.5
+ Windows 10 1809

StefanSch, 2018-10-09:
just commited a fix and opened a pull request