energia/msp430-lg-core #72: 2 UARTs + 1 SPI on MSP430FR5994

rei-vilo, 2018-10-08:
I'm trying to use 2 UARTs + 1 SPI on a MSP430FR5994.

Unfortunately, I am confused when I compare the schematics of the LaunchPad with the `pins_energia.h` header file.

* MSP430FR5994 [slau678a](http://www.ti.com/lit/ug/slau678a/slau678a.pdf) PDF page 34

> LP-pin 3 = J1.3 P6.1 MSP-pin 9
> LP-pin 4 = J1.4 P6.0 MSP-pin 8
> 
> P2.1 RX debug - ezFET MSP-pin 42
> P2.0 TX debug - ezFET MSP-pin 43

* MSP430FR5994 `pins_energia.h` header file

 ``` 
static const uint8_t DEBUG_UARTRXD = 42;  /* Receive  Data (RXD) at P3.5 */
static const uint8_t DEBUG_UARTTXD = 41;  /* Transmit Data (TXD) at P3.4 */
static const uint8_t AUX_UARTRXD = 3;     /* Receive  Data (RXD) at P4.3 */
static const uint8_t AUX_UARTTXD = 4;     /* Transmit Data (TXD) at P4.2 */
```

```
P3.5 = MSP-pin 48
P3.4 = MSP-pin 49
P4.3 = MSP-pin 34
P4.2 = MSP-pin 33
```

rei-vilo, 2018-10-08:
Basic code is

```
void setup() {
  pinMode(RED_LED, OUTPUT);
  Serial.begin(9600);
  Serial1.begin(9600);
}

void loop() {
  delay(500);
  digitalWrite(RED_LED, 1 - digitalRead(RED_LED));

  Serial.print(millis());
  Serial.println(" Serial");

  Serial1.print(millis());
  Serial1.println(" Serial1");
}
```

A logic analyser gets nothing from pins 3 / 4.

rei-vilo, 2018-10-08:
The exact same code works fine against the MSP430FR5969 LaunchPad.

StefanSch, 2018-10-09:
just commited a fix and opened a pull request

StefanSch, 2018-11-15:
merged

closed 2018-11-15 by StefanSch