energia/msp430-lg-core #75: GCC 7.3.1 — Link Fails When Using Serial

rei-vilo, 2018-10-30:
When using `Serial` with the new GCC 7.3.1, the link phase results in an error.

* Error message

> /Users/ReiVilo/Library/Energia15/packages/energia/tools/msp430-gcc/7.3.1.24/bin/msp430-elf-gcc -o /Users/ReiVilo/Library/Developer/Xcode/DerivedData/Test_New_GCC-bhrwisvezjaeunbfiqaztcfpzafx/Build/Products/Debug/embeddedcomputing.elf /Users/ReiVilo/Library/Developer/Xcode/DerivedData/Test_New_GCC-bhrwisvezjaeunbfiqaztcfpzafx/Build/Products/Debug/LocalLibrary.cpp.o /Users/ReiVilo/Library/Developer/Xcode/DerivedData/Test_New_GCC-bhrwisvezjaeunbfiqaztcfpzafx/Build/Products/Debug/main.cpp.o    /Users/ReiVilo/Library/Developer/Xcode/DerivedData/Test_New_GCC-bhrwisvezjaeunbfiqaztcfpzafx/Build/Products/Debug/embeddedcomputing.a -Os -w -fno-rtti -fno-exceptions -Wl,--gc-sections,-u,main  -mmcu=msp430fr5994 -DF_CPU= -L/Users/ReiVilo/Library/Energia15/packages/energia/tools/msp430-gcc/7.3.1.24/include -LBuilds -lm  -mlarge -mcode-region=upper -mhwmult=auto
> /Users/ReiVilo/Library/Developer/Xcode/DerivedData/Test_New_GCC-bhrwisvezjaeunbfiqaztcfpzafx/Build/Products/Debug/embeddedcomputing.a(HardwareSerial.cpp.o):(.rodata._ZTI6Stream[typeinfo for Stream]+0x0): undefined reference to 'vtable for __cxxabiv1::__si_class_type_info'
> /Users/ReiVilo/Library/Developer/Xcode/DerivedData/Test_New_GCC-bhrwisvezjaeunbfiqaztcfpzafx/Build/Products/Debug/embeddedcomputing.a(HardwareSerial.cpp.o):(.rodata._ZTI14HardwareSerial[typeinfo for HardwareSerial]+0x0): undefined reference to 'vtable for __cxxabiv1::__si_class_type_info'
> /Users/ReiVilo/Library/Developer/Xcode/DerivedData/Test_New_GCC-bhrwisvezjaeunbfiqaztcfpzafx/Build/Products/Debug/embeddedcomputing.a(Print.cpp.o):(.rodata._ZTI5Print[typeinfo for Print]+0x0): undefined reference to 'vtable for __cxxabiv1::__class_type_info'

* Minimal code

```
uint8_t myLED;

void setup()
{
    myLED = RED_LED;
    Serial.begin(9600);
    pinMode(myLED, OUTPUT);
}

void loop()
{
    Serial.println(millis());
    digitalWrite(myLED, 1 - digitalRead(myLED));
    delay(1000);
}
```

StefanSch, 2018-10-31:
you tried this on MAC - right.
Do you have any chance to also test it on a Windows machine? 
I just tried but could not see the issue.

StefanSch, 2018-10-31:
issue found and fixed - no issue on energia

closed 2018-10-31 by StefanSch

rei-vilo, 2018-10-31:
Parameter `-fno-rtti` was missing from link command.