energia/msp430-lg-core #83: attachInterrupt() doesn't work on MSP430FR5994

rei-vilo, 2019-01-01:
Consider the following minimalistic code:

```
volatile bool flag = false;

void HWI()
{
    flag = true;
}

void setup()
{
    pinMode(GREEN_LED, OUTPUT);
    digitalWrite(GREEN_LED, LOW);
    
    pinMode(PUSH1, INPUT_PULLUP);
    attachInterrupt(PUSH1, HWI, FALLING);
}

void loop()
{
    if (flag) {
        digitalWrite(GREEN_LED, HIGH);
        sleep(200);                   
        digitalWrite(GREEN_LED, LOW);
        flag = false;
    }
}
```

Pressing PUSH1 doesn't raise the interrupt and doesn't light the LED.

However, the following code polling PUSH1 works fine.

```
volatile bool flag = false;

void setup()
{
    pinMode(GREEN_LED, OUTPUT);
    digitalWrite(GREEN_LED, LOW);
    
    pinMode(PUSH1, INPUT_PULLUP);
}

void loop()
{
    flag = (digitalRead(PUSH1) == LOW);
    
    if (flag) {
        digitalWrite(GREEN_LED, HIGH);
        sleep(200);
        digitalWrite(GREEN_LED, LOW);
        flag = false;
    }
}
```

StefanSch, 2019-01-08:
fixed with pull request on branch issue_83:
https://github.com/energia/msp430-lg-core/pull/84
and merged to master.

closed 2019-01-08 by StefanSch

rei-vilo, 2019-01-08:
Thanks! It works fine now.