energia/msp430-lg-core #94: EUSCI_B0_VECTOR not declared for MSP-EXP430FR5969LP and MSP-EXP430FR6989LP

rei-vilo, 2019-02-10:
For MSP-EXP430FR5969LP and MSP-EXP430FR6989LP

> ~/Library/Energia15/packages/energia/hardware/msp430elf/2.0.4/cores/msp430/usci_isr_handler.c:198:26: error: 'EUSCI_B0_VECTOR' undeclared here (not in a function); did you mean 'USCI_B0_VECTOR'?

For MSP-EXP430FR6989LP

>~/Library/Energia15/packages/energia/hardware/msp430elf/2.0.4/cores/msp430/usci_isr_handler.c:217:26: error: 'EUSCI_B1_VECTOR' undeclared here (not in a function); did you mean 'EUSCI_B0_VECTOR'?

StefanSch, 2019-02-11:
Hi,
can you check if the version you currently i have is identical to the version on github branch SPI_rework.
I did several updates (also with suggestions from Eurol) and it think this should already be fixed there.
Thanks,
 Stefan

rei-vilo, 2019-02-11:
Thank you!

Indeed, the [SPI_rework](https://github.com/energia/msp430-lg-core/tree/SPI_rework) branch fixes the issue.

Maybe time for a new release of the MSP430ELF boards package?

closed 2019-02-11 by rei-vilo