energia/msp430-lg-core #97: Updated I²C library — Feed-back + suggestion

rei-vilo, 2019-02-12:
I've tested the updated I²C library on an MSP430FR4133 connected to an INA226 sensor. Everything when fine.

I haven't tested the library as I²C slave yet.

A nice addition would be to allow to choose between the different speeds or modes (links to www.i2c-bus.org website).

+ [Standard mode](https://www.i2c-bus.org/standard-mode/) at 100 kb/s (default)
+ [Fast mode](https://www.i2c-bus.org/fastmode/) at 400 kb/s
+ [Fast mode+](https://www.i2c-bus.org/fast-mode-plus/) at 1 Mb/s, when available

StefanSch, 2019-02-12:
Hi,
just pushed a version to the wire_update branch.
If you would like to give it a try.
Note: it uses the function prototypes from Arduino with e.g.
Wire.setClock(400000); // set 400kb/s

~ Stefan

rei-vilo, 2019-02-12:
Thanks, I'll give a try and let you know!

StefanSch, 2019-02-18:
added in wire_update branchand merged.
Note: MSP430 only supports speed up to 400k

closed 2019-02-18 by StefanSch