energia/msp430-lg-core #120: msp430-elf-gcc and DSLite need to be updated for MSP430FR2476

rei-vilo, 2019-11-07:
While compiling against the MSP430FR2476, the process raises the error

```
/msp430-elf-gcc:1:1: Device msp430fr2476 not found in devices.csv.  Check the device name is correct or obtain the latest version of the msp430-gcc-support-files archive from: 
"http://software-dl.ti.com/msp430/msp430_public_sw/mcu/msp430/MSPGCC/latest/index_FDS.html"
and place the full path to the "include" directory containing devices.csv on an include path specified with -I
```

Updating to MSP430-GCC 8.2.0.0 with the Header and Support Files from [MSP430-GCC  8_2_0_0](http://software-dl.ti.com/msp430/msp430_public_sw/mcu/msp430/MSPGCC/8_2_0_0/index_FDS.html) solved the issue.

In a similar way, upload fails. The configuration file `MSP-EXP430FR2476LP.ccxml` needs to be added to the DSLite folder.

rei-vilo, 2019-11-07:
Might be related to #74 #85 #87 #95

rei-vilo, 2019-11-21:
Use `msp430-elf-gcc 8.3.0.16` and `msp430-gcc-support-files 1.208` as previous versions have a bug —see #122.

rei-vilo, 2020-04-17:
Release `1.0.7` includes DSLite `9.3.0.1863` with support for MSP430FR2476 with `MSP-EXP430FR2476LP.ccxml`.

robertinant, 2020-04-17:
Yes, but unfortunately the header files seem to be missing from gcc for the  MSP430FR2476. I have asked @StefanSch to please look at it.

rei-vilo, 2020-04-17:
Yes, you're right for the compilers, either `msp430-gcc` or `msp430-elf-gcc`. 

Actually, I solved the issue by installing the latest version of the support files. See #123.

StefanSch, 2020-04-20:
I just did a compare of the tool files which got installed with theEnergia release E023 (where the updated compiler was part of the Energia package and the new MSP430 release 1.0.7 which maps to compiler release 4.6.6
 
There are some files missing and some patches not applied? 
How did you build the new package? Did you use the scripts I have generated?
Because this should do all the updates and patches and I this case something went wrong here.