energia/msp430-lg-core #121: Unable to load sketch to MSP430G2553

gmagannaDevelop, 2019-11-08:
I'm trying to use a 16×2 Dot matrix LCD display with my MSP430G2553 launchpad, but Energia seems to be unable to upload the sketch.

1. I'm using this version of Energia :
[image]

2. On a mid-2012 MacBook Pro running macOS Mojave.
[image]

3. I've correctly selected the board model :
[image]

4. I've tried both USB ports : 
[image]

It seems to compile successfully, but is not able to load the sketch.
[image]

I've been looking for a solution, but most solved problems regarding the error 
    
    usbutil: unable to find a device matching 0451:f432

have occurred on Windows platforms.

gmagannaDevelop, 2019-11-12:
Is there a change to get any sort of feedback ? 
I can't currently test my sketches due to this issue. 
I installed the kext as ex plained in https://energia.nu/guide/install/macos/

rei-vilo, 2019-11-12:
No problem on my side with macOS 10.15.1 _Catalina_.

Plugging-in a MSP430G2553 first generation LaunchPad and launching `ls /dev/tty.*` list the serial port `/dev/tty.uart-19FF427A4D19210C`.

```
% ls /dev/tty.*
/dev/tty.Bluetooth-Incoming-Port	/dev/tty.uart-19FF427A4D19210C
```

Similarly, both uploading with **mspdebug**...

```
/Users/ReiVilo/Library/Energia15/packages/energia/tools/mspdebug/0.24/mspdebug rf2500 --force-reset "prog /Users/ReiVilo/Library/Developer/Xcode/DerivedData/embed1-fykzaoosymfojzbkmxucgqqixhyu/Build/Products/Debug/embeddedcomputing.hex"
```

...and using `screen` for serial console...

```
% screen /dev/tty.uart-19FF427A4D19210C
```

...work fine on my side.

closed 2019-11-12 by rei-vilo

gmagannaDevelop, 2019-11-17:
I found out the selected board model was wrong, I'm terribly sorry. 
Nevertheless, I still have some problems loading my sketches :
[image]

I don't understand why I see two ports :

    $ ls /dev/tty.usb*                                
    /dev/tty.usbmodem14201 /dev/tty.usbmodem14203

rei-vilo, 2019-11-17:
See [XDS110 Debug Probe](http://www.ti.com/lit/ug/sprui94/sprui94.pdf)

> Enhanced and auxiliary debug communications
> – Support for trace capture through ARM serial wire output (SWO) – UART mode only
> – Support for UART communications to and from the target system
>

StefanSch, 2019-11-18:
Or more from the MSP430 side:
- 1st port for the debugging
- 2nd port: Backchannel Uart: can be used by the user for uart communication. See RX and TX jumpers on the header between debugger and target device.