energia/msp430-lg-core #122: "__init_array_start" symbol (msp430fr5994.ld) is not word aligned

rei-vilo, 2019-11-18:
I'm building the same project against the MSP430FR5994 using two different versions of the `msp430-elf-gcc` tool-chain by Mitto.

- Tool-chain 	msp430-elf-gcc (Mitto Systems Limited - msp430-gcc 7.3.1.24) 7.3.1

Build and link processes with no error message, application runs successfully

- Tool-chain 	msp430-elf-gcc (Mitto Systems Limited - msp430-gcc 8.2.0.52) 8.2.0

Build and link processes with one warning message, but the application freezes at start.

> /Users/ReiVilo/Library/Energia15/packages/energia/tools/msp430-elf-gcc/8.2.0.52/bin/../lib/gcc/msp430-elf/8.2.0/../../../../msp430-elf/bin/ld:1:1: "__init_array_start" symbol (msp430fr5994.ld) is not word aligned

Could you please confirm this issue and, if needed, escalate it to Mitto Systems Limited?

thank you!

rei-vilo, 2019-11-18:
I haven't tried other combinations, like 

+ `msp430-elf-gcc 7.3.1` and `msp430-gcc-support-files 1.207`, or
+ `msp430-elf-gcc 8.2.0` and `msp430-gcc-support-files 1.206`.

StefanSch, 2019-11-21:
just tried this config:
GCC_VER="8.3.0.16"
MSPSUPPORT_VER="1.208"
but could not see an issue

rei-vilo, 2019-11-21:
I’ll do the updates and try again.

rei-vilo, 2019-11-21:
According to the `Revisions_Header.txt`:

> Build 1.208 (GCC)
> =================
> 08/29/2019
> 
> 
> Bug fixes:
> ----------
> - Fixed GNU compiler build options
> - Added missing ALIGN directives for .preinit/init/fini_array sections in GCC
>   linker command files
>

rei-vilo, 2019-11-21:
Indeed, `msp430-elf-gcc 8.3.0.16` and `msp430-gcc-support-files 1.208` work fine.  

No more warning on build, and the application runs as expected.

So please include those versions with the next release of the MSP430 boards package.

closed 2019-11-21 by rei-vilo