energia/msp430-lg-core #123: Why aren't the msp430  gcc support files 1.208 used?

trick2011, 2020-01-13:
I just noticed that the support files used for the MCUs are version 1.159. The most recent version seems to be 1.208. So I'm wondering why the latest version isn't in use?

I know about performance problems with the newer compilers after gcc 4.6 but this shouldn't interfere with the support files, right?

rei-vilo, 2020-01-13:
You can download and install the support files independently. 

Please refer to [GCC - Open Source Compiler for MSP Microcontrollers](http://www.ti.com/tool/download/MSP430-GCC-OPENSOURCE) for more information.

trick2011, 2020-01-13:
@rei-vilo yeah of course but that wasn't the question. Is there a specific reason why this repo hasn't upstreamed the new library into the repository?

rei-vilo, 2020-01-13:
Have you gone through all the different branches?

trick2011, 2020-01-14:
Yeah and none of them upstream the new support library. So the question remains: is there a **problem** with the new support library?

StefanSch, 2020-01-14:
The new support library and header files are generated for the new compiler.
Energia uses still the old compiler version based on 4.6.0
So converting the header files is quite an effort.
But we are working on releasing an Energia version which uses the latest compiler version release for MSP430.
If need you can also build the package on your own. All generation files and scripts are push on the Energia github for the MSP430

trick2011, 2020-01-14:
Quite logical that they target the new compiler but do they use functions not available in the old compiler? It seems to me that they would target C99 at most for this?

@StefanSch I tried the new_compiler branch but am having difficulties getting to download & unrar/unzip

StefanSch, 2020-01-14:
The latest version of the script is already merged to the master branch - so run it from there.
cd into base folder of the project and then ./extras/make_mito_release.sh