energia/msp430-lg-core #128: Install this package to use your Energia/MSP430 Tool board

rei-vilo, 2020-04-17:
On macOS, Energia displays this message at launch.

[image]

Clicking on the link opens the **Boards manager** but goes nowhere.

[image]

## References

+ Energia 1.8.11E23
+ MSP430 core 1.0.7

rei-vilo, 2020-05-02:
Same issue on Windows.

[image]

Clicking on the link opens the **Boards Manager** with an empty list.

[image]

jahagirdar, 2021-02-09:
Same Issue on Linux.