energia/msp430-lg-core #129: DSLite 9.3.0.1863 macOS notarisation issue

rei-vilo, 2020-04-17:
When uploading with DSLite `9.3.0.1863` provided with MSP430 core `1.0.7`, macOS raises error messages about notarisation.

“libfsutil.dylib” cannot be opened because the developer cannot be verified.

[image]

The workaround requires opening each dynamic libraries of DSLite and authorising them manually.

Other cores like [Tiva C core 1.0.4](https://github.com/energia/tivac-core/releases/tag/v1.0.4) might be affected as well.

rei-vilo, 2020-04-17:
Another workaround is provided at https://www.imore.com/how-open-apps-anywhere-macos-catalina-and-mojave.

robertinant, 2020-04-17:
I did not see that on my Mac. Will check my other Mac to verify.

robertinant, 2020-04-17:
Do you see anything similar with ticloud?

rei-vilo, 2020-04-17:
I managed to [authorise DSLite](https://github.com/energia/msp430-lg-core/issues/129#issuecomment-615367286) 42 minutes ago so using TI Cloud now won't raise the message again.