energia/msp430-lg-core #131: Release 2.0.9 with msp430-elf-gcc 9.2.0.50 fails to build the `Blink` example against the MSP430FR5994.

rei-vilo, 2020-09-16:
Release 2.0.9 with msp430-elf-gcc 9.2.0.50 fails to build the `Blink` example against the MSP430FR5994.

```
/Users/USER/Library/Energia15/packages/energia/tools/msp430-elf-gcc/9.2.0.50/bin/msp430-elf-gcc -c -g -Os -w -ffunction-sections -fdata-sections -MMD -mmcu=msp430fr5994 -DF_CPU=16000000L -DARDUINO=10811 -DENERGIA=10811 -DENERGIA_MSP_EXP430FR5994LP -DENERGIA_ARCH_MSP430ELF -I/Users/USER/Library/Energia15/packages/energia/tools/msp430-elf-gcc/9.2.0.50/include -I/Users/USER/Library/Energia15/packages/energia/tools/msp430-elf-gcc/9.2.0.50/msp430-gcc-support-files/include -mlarge -mcode-region=either -mhwmult=auto -I/Users/USER/Library/Energia15/packages/energia/hardware/msp430elf/2.0.9/cores/msp430 -I/Users/USER/Library/Energia15/packages/energia/hardware/msp430elf/2.0.9/variants/MSP-EXP430FR5994LP /Users/USER/Library/Energia15/packages/energia/hardware/msp430elf/2.0.9/cores/msp430/wiring_pulse.c -o /var/folders/wn/n7qqb8ss0k3bvpqwfcdwp_7r0000gn/T/arduino_build_223338/core/wiring_pulse.c.o
In file included from /Users/USER/Library/Energia15/packages/energia/hardware/msp430elf/2.0.9/cores/msp430/Energia.h:366,
                 from /Users/USER/Library/Energia15/packages/energia/hardware/msp430elf/2.0.9/cores/msp430/wiring_private.h:37,
                 from /Users/USER/Library/Energia15/packages/energia/hardware/msp430elf/2.0.9/cores/msp430/wiring_digital.c:33:
/Users/USER/Library/Energia15/packages/energia/hardware/msp430elf/2.0.9/variants/MSP-EXP430FR5994LP/pins_energia.h:246:23: error: 'P2_BASE' undeclared here (not in a function); did you mean 'P1_BASE'?
  246 |   #define P2_X_BASE   P2_BASE
      |                       ^~~~~~~
/Users/USER/Library/Energia15/packages/energia/hardware/msp430elf/2.0.9/variants/MSP-EXP430FR5994LP/pins_energia.h:262:16: note: in expansion of macro 'P2_X_BASE'
  262 |    (uint16_t) (P2_X_BASE+OFS_P2IN),
      |                ^~~~~~~~~
/Users/USER/Library/Energia15/packages/energia/hardware/msp430elf/2.0.9/variants/MSP-EXP430FR5994LP/pins_energia.h:295:4: error: initializer element is not constant
  295 |    (uint16_t) (P2_X_BASE+OFS_P2OUT),
      |    ^
/Users/USER/Library/Energia15/packages/energia/hardware/msp430elf/2.0.9/variants/MSP-EXP430FR5994LP/pins_energia.h:295:4: note: (near initialization for 'port_to_output[2]')
/Users/USER/Library/Energia15/packages/energia/hardware/msp430elf/2.0.9/variants/MSP-EXP430FR5994LP/pins_energia.h:328:4: error: initializer element is not constant
  328 |    (uint16_t) (P2_X_BASE+OFS_P2DIR),
      |    ^
/Users/USER/Library/Energia15/packages/energia/hardware/msp430elf/2.0.9/variants/MSP-EXP430FR5994LP/pins_energia.h:328:4: note: (near initialization for 'port_to_dir[2]')
/Users/USER/Library/Energia15/packages/energia/hardware/msp430elf/2.0.9/variants/MSP-EXP430FR5994LP/pins_energia.h:361:5: error: initializer element is not constant
  361 |     (uint16_t) (P2_X_BASE+OFS_P2REN),
      |     ^
/Users/USER/Library/Energia15/packages/energia/hardware/msp430elf/2.0.9/variants/MSP-EXP430FR5994LP/pins_energia.h:361:5: note: (near initialization for 'port_to_ren[2]')
/Users/USER/Library/Energia15/packages/energia/hardware/msp430elf/2.0.9/variants/MSP-EXP430FR5994LP/pins_energia.h:394:5: error: initializer element is not constant
  394 |     (uint16_t) (P2_X_BASE+OFS_P2SEL0),
      |     ^
/Users/USER/Library/Energia15/packages/energia/hardware/msp430elf/2.0.9/variants/MSP-EXP430FR5994LP/pins_energia.h:394:5: note: (near initialization for 'port_to_sel0[2]')
/Users/USER/Library/Energia15/packages/energia/hardware/msp430elf/2.0.9/variants/MSP-EXP430FR5994LP/pins_energia.h:427:5: error: initializer element is not constant
  427 |     (uint16_t) (P2_X_BASE+OFS_P2SEL1),
      |     ^
/Users/USER/Library/Energia15/packages/energia/hardware/msp430elf/2.0.9/variants/MSP-EXP430FR5994LP/pins_energia.h:427:5: note: (near initialization for 'port_to_sel1[2]')
/Users/USER/Library/Energia15/packages/energia/hardware/msp430elf/2.0.9/variants/MSP-EXP430FR5994LP/pins_energia.h:516:4: error: 'P6' undeclared here (not in a function); did you mean 'A6'?
  516 |    P6,        /* 3  - P6.1 */
      |    ^~
      |    A6
/Users/USER/Library/Energia15/packages/energia/hardware/msp430elf/2.0.9/variants/MSP-EXP430FR5994LP/pins_energia.h:516:4: error: initializer element is not constant
/Users/USER/Library/Energia15/packages/energia/hardware/msp430elf/2.0.9/variants/MSP-EXP430FR5994LP/pins_energia.h:516:4: note: (near initialization for 'digital_pin_to_port[3]')
/Users/USER/Library/Energia15/packages/energia/hardware/msp430elf/2.0.9/variants/MSP-EXP430FR5994LP/pins_energia.h:517:4: error: initializer element is not constant
  517 |    P6,        /* 4  - P6.0 */
      |    ^~
/Users/USER/Library/Energia15/packages/energia/hardware/msp430elf/2.0.9/variants/MSP-EXP430FR5994LP/pins_energia.h:517:4: note: (near initialization for 'digital_pin_to_port[4]')
/Users/USER/Library/Energia15/packages/energia/hardware/msp430elf/2.0.9/variants/MSP-EXP430FR5994LP/pins_energia.h:518:4: error: initializer element is not constant
  518 |    P6,        /* 5  - P6.2 */
      |    ^~
/Users/USER/Library/Energia15/packages/energia/hardware/msp430elf/2.0.9/variants/MSP-EXP430FR5994LP/pins_energia.h:518:4: note: (near initialization for 'digital_pin_to_port[5]')
/Users/USER/Library/Energia15/packages/energia/hardware/msp430elf/2.0.9/cores/msp430/WInterrupts.c: In function 'attachInterrupt':
/Users/USER/Library/Energia15/packages/energia/hardware/msp430elf/2.0.9/cores/msp430/WInterrupts.c:132:10: error: 'P3' undeclared (first use in this function); did you mean 'A3'?
  132 |     case P3:
      |          ^~
      |          A3
/Users/USER/Library/Energia15/packages/energia/hardware/msp430elf/2.0.9/cores/msp430/WInterrupts.c:132:10: note: each undeclared identifier is reported only once for each function it appears in
/Users/USER/Library/Energia15/packages/energia/hardware/msp430elf/2.0.9/cores/msp430/WInterrupts.c:145:10: error: 'P4' undeclared (first use in this function); did you mean 'A4'?
  145 |     case P4:
      |          ^~
      |          A4
/Users/USER/Library/Energia15/packages/energia/hardware/msp430elf/2.0.9/variants/MSP-EXP430FR5994LP/pins_energia.h:520:4: error: 'P5' undeclared here (not in a function); did you mean 'A5'?
  520 |    P5,        /* 7  - P5.2 */
      |    ^~
      |    A5
/Users/USER/Library/Energia15/packages/energia/hardware/msp430elf/2.0.9/variants/MSP-EXP430FR5994LP/pins_energia.h:520:4: error: initializer element is not constant
/Users/USER/Library/Energia15/packages/energia/hardware/msp430elf/2.0.9/variants/MSP-EXP430FR5994LP/pins_energia.h:520:4: note: (near initialization for 'digital_pin_to_port[7]')
/Users/USER/Library/Energia15/packages/energia/hardware/msp430elf/2.0.9/variants/MSP-EXP430FR5994LP/pins_energia.h:521:4: error: initializer element is not constant
  521 |    P6,        /* 8  - P6.3 */
      |    ^~
/Users/USER/Library/Energia15/packages/energia/hardware/msp430elf/2.0.9/variants/MSP-EXP430FR5994LP/pins_energia.h:521:4: note: (near initialization for 'digital_pin_to_port[8]')
/Users/USER/Library/Energia15/packages/energia/hardware/msp430elf/2.0.9/cores/msp430/WInterrupts.c:158:10: error: 'P5' undeclared (first use in this function); did you mean 'A5'?
  158 |     case P5:
      |          ^~
[... 124 lines cut ...]
/Users/USER/Library/Energia15/packages/energia/hardware/msp430elf/2.0.9/variants/MSP-EXP430FR5994LP/pins_energia.h:553:4: error: initializer element is not constant
  553 |    P3,        /* 40 - P3.7 */
      |    ^~
/Users/USER/Library/Energia15/packages/energia/hardware/msp430elf/2.0.9/variants/MSP-EXP430FR5994LP/pins_energia.h:553:4: note: (near initialization for 'digital_pin_to_port[40]')
/Users/USER/Library/Energia15/packages/energia/hardware/msp430elf/2.0.9/variants/MSP-EXP430FR5994LP/pins_energia.h:558:4: error: initializer element is not constant
  558 |    P5,        /* 45 - P5.6 */
      |    ^~
/Users/USER/Library/Energia15/packages/energia/hardware/msp430elf/2.0.9/variants/MSP-EXP430FR5994LP/pins_energia.h:558:4: note: (near initialization for 'digital_pin_to_port[45]')
/Users/USER/Library/Energia15/packages/energia/hardware/msp430elf/2.0.9/variants/MSP-EXP430FR5994LP/pins_energia.h:559:4: error: initializer element is not constant
  559 |    P5,        /* 46 - P5.5 */
      |    ^~
[... cut ...]
/Users/USER/Library/Energia15/packages/energia/hardware/msp430elf/2.0.9/variants/MSP-EXP430FR5994LP/pins_energia.h:564:4: error: initializer element is not constant
  564 |    P7,        /* 51 - P7.2 */
      |    ^~
/Users/USER/Library/Energia15/packages/energia/hardware/msp430elf/2.0.9/variants/MSP-EXP430FR5994LP/pins_energia.h:564:4: note: (near initialization for 'digital_pin_to_port[51]')
exit status 1
Error compiling for board MSP-EXP430FR5994LP.
```

Using legacy release 1.0.7 with msp430-gcc 4.6.6 succeeds.

rei-vilo, 2020-09-16:
@StefanSch Using previous releases 8.3.0.16 or 8.3.1.25 of msp430-elf-gcc doesn't raise the issue.

StefanSch, 2020-09-17:
Thanks for finding. 
Unfortunately the header files in the latest release from TI got corrupted or some information got removed.
I just made another release with the previous version of header files 1.209
Also the previous package of energia with elf compiler bases on compiler version 8 is working (energia version 2.0.9).
So now the energia version 2.0.10 has compiler version 9.x but uses the previous header package 1.209

rei-vilo, 2020-09-17:
@StefanSch  You're welcome!

Release `2.0.10` fixes the issue.

I'm closing the ticket.

closed 2021-06-01 by rei-vilo