energia/msp430-lg-core #132: Release `2.0.10` with `-DF_CPU=16000000L` and FRAM

rei-vilo, 2020-09-17:
Release `2.0.10` mentions `-DF_CPU=16000000L`.

Since FRAM is limited to 8 MHz, is the 16 MHz clock-speed compatible with FRAM?

StefanSch, 2020-09-21:
16 MHz are allowed if for the FRAM the waitstates get enabled :

#if F_CPU >= 16000000L
    FRCTL0 = FWPW | NACCESS_1;     // add 1 waitstaite
#elif F_CPU >= 12000000L
    FRCTL0 = FWPW | NACCESS_1;     // add 1 waitstaite
#elif F_CPU >= 8000000L
#elif F_CPU >= 1000000L
#else

rei-vilo, 2020-09-21:
Thanks!  

So the question is now: does Energia include that configuration? It seems so.

https://github.com/energia/msp430-lg-core/blob/90728451fcb1d6615c1d2ac5161de643feae6027/cores/msp430/wiring.c#L319

I remember having faced issues with the MSP430FR5994 FRAM on my [FRAM-based E-Paper Screen Controller](https://embeddedcomputing.weebly.com/fram-based-e-paper-screen-controller.html).

rei-vilo, 2021-04-12:
Fixed with release 2.0.10 MSP430 ELF-GCC.

closed 2021-04-12 by rei-vilo