energia/msp430-lg-core #135: Sparkfun icm 20498 arduino libraries migration problem for msp430fr5994

tarunrtes, 2021-09-30:
E:\Tarun Karak\Energia\Energia_software\energia-1.8.7E21\libraries\SparkFun_9DoF_IMU_Breakout___ICM_20948___Arduino_Library-1.2.7\src/util/ICM_20948_DMP.h:548:17: error: expected unqualified-id before numeric constant

E:\Tarun Karak\Energia\Energia_software\energia-1.8.7E21\libraries\SparkFun_9DoF_IMU_Breakout___ICM_20948___Arduino_Library-1.2.7\src/util/ICM_20948_DMP.h:548:17: error: expected ')' before numeric constant

E:\Tarun Karak\Energia\Energia_software\energia-1.8.7E21\libraries\SparkFun_9DoF_IMU_Breakout___ICM_20948___Arduino_Library-1.2.7\src/util/ICM_20948_DMP.h:558:17: error: expected unqualified-id before numeric constant

E:\Tarun Karak\Energia\Energia_software\energia-1.8.7E21\libraries\SparkFun_9DoF_IMU_Breakout___ICM_20948___Arduino_Library-1.2.7\src/util/ICM_20948_DMP.h:558:17: error: expected ')' before numeric constant

E:\Tarun Karak\Energia\Energia_software\energia-1.8.7E21\libraries\SparkFun_9DoF_IMU_Breakout___ICM_20948___Arduino_Library-1.2.7\src/util/ICM_20948_DMP.h:571:17: error: expected unqualified-id before numeric constant

E:\Tarun Karak\Energia\Energia_software\energia-1.8.7E21\libraries\SparkFun_9DoF_IMU_Breakout___ICM_20948___Arduino_Library-1.2.7\src/util/ICM_20948_DMP.h:571:17: error: expected ')' before numeric constant

E:\Tarun Karak\Energia\Energia_software\energia-1.8.7E21\libraries\SparkFun_9DoF_IMU_Breakout___ICM_20948___Arduino_Library-1.2.7\src/util/ICM_20948_DMP.h:634:17: error: expected unqualified-id before numeric constant

E:\Tarun Karak\Energia\Energia_software\energia-1.8.7E21\libraries\SparkFun_9DoF_IMU_Breakout___ICM_20948___Arduino_Library-1.2.7\src/util/ICM_20948_DMP.h:634:17: error: expected ')' before numeric constant

E:\Tarun Karak\Energia\Energia_software\energia-1.8.7E21\libraries\SparkFun_9DoF_IMU_Breakout___ICM_20948___Arduino_Library-1.2.7\src/util/ICM_20948_DMP.h:644:17: error: expected unqualified-id before numeric constant

E:\Tarun Karak\Energia\Energia_software\energia-1.8.7E21\libraries\SparkFun_9DoF_IMU_Breakout___ICM_20948___Arduino_Library-1.2.7\src/util/ICM_20948_DMP.h:644:17: error: expected ')' before numeric constant

In file included from C:\Users\TARUN KARAK\Documents\Energia\libraries\SparkFun_9DoF_IMU_Breakout___ICM_20948___Arduino_Library-1.2.7\examples\Arduino\Example1_Basics\Example1_Basics.ino:12:0:

E:\Tarun Karak\Energia\Energia_software\energia-1.8.7E21\libraries\SparkFun_9DoF_IMU_Breakout___ICM_20948___Arduino_Library-1.2.7\src/ICM_20948.h:24:22: error: ISO C++ forbids initialization of member '_printDebug' [-fpermissive]

E:\Tarun Karak\Energia\Energia_software\energia-1.8.7E21\libraries\SparkFun_9DoF_IMU_Breakout___ICM_20948___Arduino_Library-1.2.7\src/ICM_20948.h:24:22: error: making '_printDebug' static [-fpermissive]

E:\Tarun Karak\Energia\Energia_software\energia-1.8.7E21\libraries\SparkFun_9DoF_IMU_Breakout___ICM_20948___Arduino_Library-1.2.7\src/ICM_20948.h:24:22: error: ISO C++ forbids in-class initialization of non-const static member '_printDebug'

E:\Tarun Karak\Energia\Energia_software\energia-1.8.7E21\libraries\SparkFun_9DoF_IMU_Breakout___ICM_20948___Arduino_Library-1.2.7\src/ICM_20948.h:26:43: error: ISO C++ forbids initialization of member 'MAX_MAGNETOMETER_STARTS' [-fpermissive]

E:\Tarun Karak\Energia\Energia_software\energia-1.8.7E21\libraries\SparkFun_9DoF_IMU_Breakout___ICM_20948___Arduino_Library-1.2.7\src/ICM_20948.h:26:43: error: making 'MAX_MAGNETOMETER_STARTS' static [-fpermissive]

exit status 1
Error compiling for board MSP-EXP430FR5994LP.

I am migrating a project from Arduino to energia. Why I am getting these errors? please help.

StefanSch, 2021-10-01:
the official version of Energia for the MSP430 is using a quite old compiler and it might be an issue due to the compatibility int he compiler versions.
There is a beta version available with the newer compiler version.
You can import MSP430 with the newer compiler into Energia (or Arduino).

Open the menu: Preferences | Additional Board Manager URLs
Just add 
http://s3.amazonaws.com/energiaUS/packages/package_msp430_elf_GCC_index.json
Open the Board Manager and install the MSP430 elf-GCC package.

Then you also find a elf-GCC section in the board selection menu.

tarunrtes, 2021-10-01:
Dear sir,
I have done everything as you suggested. But didn't get the MSP430 elf-GCC beta version. So I installed 2.0.10 and get these errors. Please help me out.
msp430-elf-gcc: fatal error: switch 'KARAK/AppData/Local/Energia15/packages/energia/tools/msp430-elf-gcc/9.2.0.50/msp430-gcc-support-files/include/devices.csv' does not start with '-'
compilation terminated.

exit status 1
Error compiling for board MSP-EXP430FR5994LP.

tarunrtes, 2021-10-04:
Dear sir, please provide MSP430 elf-GCC beta version. I have gone through your steps but didn't succeed to compile the program.

Thanks & Regards
Tarun Karak

StefanSch, 2021-10-04:
Hi, it is right to install 2.0.10 - this is still handled as beta and therefore not the the standard release available. So only by adding the additional install path.
Not sure about your issue, i have not seen this so far.
Can you provide more details: operating system, full log message also set Compiler warnings to all in the preferences.
Can you try with the simple blink LED demo?

tarunrtes, 2021-10-04:
Thanks for your response. Yes, I have tried blink-program using   MSP430
elf-GCC 2.0.10 but am not able to run the program and it is showing the
same errors. But when selecting normal msp430FR5994 (not the elf-gcc) from
the board manager I was able to run and upload the programme successfully.
Please help me to run sparkfun icm20948 arduino library.

E:\Tarun Karak\Energia\Energia_software\energia-1.8.10E23\arduino-builder
-dump-prefs -logger=machine -hardware E:\Tarun
Karak\Energia\Energia_software\energia-1.8.10E23\hardware -hardware
C:\Users\TARUN KARAK\AppData\Local\Energia15\packages -tools E:\Tarun
Karak\Energia\Energia_software\energia-1.8.10E23\tools-builder -tools
E:\Tarun
Karak\Energia\Energia_software\energia-1.8.10E23\hardware\tools\msp430
-tools C:\Users\TARUN KARAK\AppData\Local\Energia15\packages
-built-in-libraries E:\Tarun
Karak\Energia\Energia_software\energia-1.8.10E23\libraries -libraries
C:\Users\TARUN KARAK\Documents\Energia\libraries
-fqbn=energia:msp430elf:MSP-EXP430FR5994LP -ide-version=10811 -build-path
C:\Users\TARUNK~1\AppData\Local\Temp\arduino_build_228958 -warnings=all
-build-cache C:\Users\TARUNK~1\AppData\Local\Temp\arduino_cache_755109
-prefs=build.warn_data_percentage=75
-prefs=runtime.tools.mspdebug.path=C:\Users\TARUN
KARAK\AppData\Local\Energia15\packages\energia\tools\mspdebug\0.24
-prefs=runtime.tools.mspdebug-0.24.path=C:\Users\TARUN
KARAK\AppData\Local\Energia15\packages\energia\tools\mspdebug\0.24
-prefs=runtime.tools.dslite.path=C:\Users\TARUN
KARAK\AppData\Local\Energia15\packages\energia\tools\dslite\9.3.0.1863
-prefs=runtime.tools.dslite-9.3.0.1863.path=C:\Users\TARUN
KARAK\AppData\Local\Energia15\packages\energia\tools\dslite\9.3.0.1863
-prefs=runtime.tools.msp430-elf-gcc.path=C:\Users\TARUN
KARAK\AppData\Local\Energia15\packages\energia\tools\msp430-elf-gcc\9.2.0.50
-prefs=runtime.tools.msp430-elf-gcc-9.2.0.50.path=C:\Users\TARUN
KARAK\AppData\Local\Energia15\packages\energia\tools\msp430-elf-gcc\9.2.0.50
-prefs=runtime.tools.ino2cpp.path=C:\Users\TARUN
KARAK\AppData\Local\Energia15\packages\energia\tools\ino2cpp\1.0.4
-prefs=runtime.tools.ino2cpp-1.0.4.path=C:\Users\TARUN
KARAK\AppData\Local\Energia15\packages\energia\tools\ino2cpp\1.0.4 -verbose
C:\Users\TARUN
KARAK\Documents\Energia\libraries\SparkFun_9DoF_IMU_Breakout___ICM_20948___Arduino_Library-1.2.7\examples\Arduino\Example1_Basics\Example1_Basics.ino
E:\Tarun Karak\Energia\Energia_software\energia-1.8.10E23\arduino-builder
-compile -logger=machine -hardware E:\Tarun
Karak\Energia\Energia_software\energia-1.8.10E23\hardware -hardware
C:\Users\TARUN KARAK\AppData\Local\Energia15\packages -tools E:\Tarun
Karak\Energia\Energia_software\energia-1.8.10E23\tools-builder -tools
E:\Tarun
Karak\Energia\Energia_software\energia-1.8.10E23\hardware\tools\msp430
-tools C:\Users\TARUN KARAK\AppData\Local\Energia15\packages
-built-in-libraries E:\Tarun
Karak\Energia\Energia_software\energia-1.8.10E23\libraries -libraries
C:\Users\TARUN KARAK\Documents\Energia\libraries
-fqbn=energia:msp430elf:MSP-EXP430FR5994LP -ide-version=10811 -build-path
C:\Users\TARUNK~1\AppData\Local\Temp\arduino_build_228958 -warnings=all
-build-cache C:\Users\TARUNK~1\AppData\Local\Temp\arduino_cache_755109
-prefs=build.warn_data_percentage=75
-prefs=runtime.tools.mspdebug.path=C:\Users\TARUN
KARAK\AppData\Local\Energia15\packages\energia\tools\mspdebug\0.24
-prefs=runtime.tools.mspdebug-0.24.path=C:\Users\TARUN
KARAK\AppData\Local\Energia15\packages\energia\tools\mspdebug\0.24
-prefs=runtime.tools.dslite.path=C:\Users\TARUN
KARAK\AppData\Local\Energia15\packages\energia\tools\dslite\9.3.0.1863
-prefs=runtime.tools.dslite-9.3.0.1863.path=C:\Users\TARUN
KARAK\AppData\Local\Energia15\packages\energia\tools\dslite\9.3.0.1863
-prefs=runtime.tools.msp430-elf-gcc.path=C:\Users\TARUN
KARAK\AppData\Local\Energia15\packages\energia\tools\msp430-elf-gcc\9.2.0.50
-prefs=runtime.tools.msp430-elf-gcc-9.2.0.50.path=C:\Users\TARUN
KARAK\AppData\Local\Energia15\packages\energia\tools\msp430-elf-gcc\9.2.0.50
-prefs=runtime.tools.ino2cpp.path=C:\Users\TARUN
KARAK\AppData\Local\Energia15\packages\energia\tools\ino2cpp\1.0.4
-prefs=runtime.tools.ino2cpp-1.0.4.path=C:\Users\TARUN
KARAK\AppData\Local\Energia15\packages\energia\tools\ino2cpp\1.0.4 -verbose
[... 14 lines cut ...]
KARAK\\AppData\\Local\\Energia15\\packages\\energia\\tools\\msp430-elf-gcc\\
9.2.0.50/msp430-gcc-support-files/include" -mlarge -mcode-region=either
-mhwmult=auto "-IC:\\Users\\TARUN
KARAK\\AppData\\Local\\Energia15\\packages\\energia\\hardware\\msp430elf\\2.0.10\\cores\\msp430"
"-IC:\\Users\\TARUN
KARAK\\AppData\\Local\\Energia15\\packages\\energia\\hardware\\msp430elf\\2.0.10\\variants\\MSP-EXP430FR5994LP"
"C:\\Users\\TARUNK~1\\AppData\\Local\\Temp\\arduino_build_228958\\sketch\\Example1_Basics.ino.cpp"
-o nul
Alternatives for switch
'KARAK/AppData/Local/Energia15/packages/energia/tools/msp430-elf-gcc/
9.2.0.50/msp430-gcc-support-files/include/devices.csv' does not start with
'-': []
ResolveLibrary(switch
'KARAK/AppData/Local/Energia15/packages/energia/tools/msp430-elf-gcc/
9.2.0.50/msp430-gcc-support-files/include/devices.csv' does not start with
'-')
  -> candidates: []
msp430-elf-gcc: fatal error: switch
'KARAK/AppData/Local/Energia15/packages/energia/tools/msp430-elf-gcc/
9.2.0.50/msp430-gcc-support-files/include/devices.csv' does not start with
'-'
compilation terminated.

exit status 1
Error compiling for board MSP-EXP430FR5994LP.

StefanSch, 2021-10-06:
This is really strange.
Can you make a copy of your file
'KARAK/AppData/Local/Energia15/packages/energia/tools/msp430-elf-gcc/
9.2.0.50/msp430-gcc-support-files/include/devices.csv'

and replace it with the attached one.
(If this does fix the problem please send me send the original file.)

[devices.csv](https://github.com/energia/msp430-lg-core/files/7291406/devices.csv)

tarunrtes, 2021-10-06:
Dear sir,
Now I am able to run basic Blink project and uploaded successfully but
still get the error message for Sparkfun icm20948 libraries. Please see the
errors.

In file included from
C:\Users\HP\AppData\Local\Energia15\packages\energia\tools\msp430-elf-gcc\
9.2.0.50/msp430-gcc-support-files/include/msp430.h:1552,
                 from
C:\Users\HP\AppData\Local\Energia15\packages\energia\hardware\msp430elf\2.0.10\cores\msp430/Energia.h:4,
                 from
C:\Users\HP\AppData\Local\Energia15\packages\energia\hardware\msp430elf\2.0.10\cores\msp430/Arduino.h:3,
                 from sketch\Example1_Basics.ino.cpp:1:
C:\Users\HP\AppData\Local\Energia15\packages\energia\tools\msp430-elf-gcc\
9.2.0.50/msp430-gcc-support-files/include/msp430fr5994.h:8774:43: error:
expected unqualified-id before numeric constant
 8774 | #define ID                               (0x00c0)        /* Input
divider */
      |                                           ^~~~~~
E:\energia-1.8.10E23-windows\energia-1.8.10E23\libraries\SparkFun_9DoF_IMU_Breakout___ICM_20948___Arduino_Library-1.2.7\src/util/ICM_20948_REGISTERS.h:761:13:
note: in expansion of macro 'ID'
  761 |     uint8_t ID : 7;
      |             ^~
C:\Users\HP\AppData\Local\Energia15\packages\energia\tools\msp430-elf-gcc\
9.2.0.50/msp430-gcc-support-files/include/msp430fr5994.h:8774:43: error:
expected ')' before numeric constant
 8774 | #define ID                               (0x00c0)        /* Input
divider */
      |                                          ~^~~~~~
E:\energia-1.8.10E23-windows\energia-1.8.10E23\libraries\SparkFun_9DoF_IMU_Breakout___ICM_20948___Arduino_Library-1.2.7\src/util/ICM_20948_REGISTERS.h:761:13:
note: in expansion of macro 'ID'
  761 |     uint8_t ID : 7;
      |             ^~
C:\Users\HP\AppData\Local\Energia15\packages\energia\tools\msp430-elf-gcc\
9.2.0.50/msp430-gcc-support-files/include/msp430fr5994.h:92:30: error:
expected unqualified-id before numeric constant
   92 | #define Z                   (0x0002)
      |                              ^~~~~~
E:\energia-1.8.10E23-windows\energia-1.8.10E23\libraries\SparkFun_9DoF_IMU_Breakout___ICM_20948___Arduino_Library-1.2.7\src/util/ICM_20948_DMP.h:548:17:
note: in expansion of macro 'Z'
  548 |         int16_t Z;
      |                 ^
C:\Users\HP\AppData\Local\Energia15\packages\energia\tools\msp430-elf-gcc\
9.2.0.50/msp430-gcc-support-files/include/msp430fr5994.h:92:30: error:
expected ')' before numeric constant
   92 | #define Z                   (0x0002)
      |                             ~^~~~~~
E:\energia-1.8.10E23-windows\energia-1.8.10E23\libraries\SparkFun_9DoF_IMU_Breakout___ICM_20948___Arduino_Library-1.2.7\src/util/ICM_20948_DMP.h:548:17:
note: in expansion of macro 'Z'
  548 |         int16_t Z;
      |                 ^
C:\Users\HP\AppData\Local\Energia15\packages\energia\tools\msp430-elf-gcc\
9.2.0.50/msp430-gcc-support-files/include/msp430fr5994.h:92:30: error:
expected unqualified-id before numeric constant
   92 | #define Z                   (0x0002)
      |                              ^~~~~~
E:\energia-1.8.10E23-windows\energia-1.8.10E23\libraries\SparkFun_9DoF_IMU_Breakout___ICM_20948___Arduino_Library-1.2.7\src/util/ICM_20948_DMP.h:558:17:
note: in expansion of macro 'Z'
  558 |         int16_t Z;
      |                 ^
C:\Users\HP\AppData\Local\Energia15\packages\energia\tools\msp430-elf-gcc\
9.2.0.50/msp430-gcc-support-files/include/msp430fr5994.h:92:30: error:
expected ')' before numeric constant
   92 | #define Z                   (0x0002)
      |                             ~^~~~~~
E:\energia-1.8.10E23-windows\energia-1.8.10E23\libraries\SparkFun_9DoF_IMU_Breakout___ICM_20948___Arduino_Library-1.2.7\src/util/ICM_20948_DMP.h:558:17:
note: in expansion of macro 'Z'
  558 |         int16_t Z;
      |                 ^
C:\Users\HP\AppData\Local\Energia15\packages\energia\tools\msp430-elf-gcc\
[... 42 lines cut ...]
  634 |         int32_t Z;
      |                 ^
C:\Users\HP\AppData\Local\Energia15\packages\energia\tools\msp430-elf-gcc\
9.2.0.50/msp430-gcc-support-files/include/msp430fr5994.h:92:30: error:
expected unqualified-id before numeric constant
   92 | #define Z                   (0x0002)
      |                              ^~~~~~
E:\energia-1.8.10E23-windows\energia-1.8.10E23\libraries\SparkFun_9DoF_IMU_Breakout___ICM_20948___Arduino_Library-1.2.7\src/util/ICM_20948_DMP.h:644:17:
note: in expansion of macro 'Z'
  644 |         int32_t Z;
      |                 ^
C:\Users\HP\AppData\Local\Energia15\packages\energia\tools\msp430-elf-gcc\
9.2.0.50/msp430-gcc-support-files/include/msp430fr5994.h:92:30: error:
expected ')' before numeric constant
   92 | #define Z                   (0x0002)
      |                             ~^~~~~~
E:\energia-1.8.10E23-windows\energia-1.8.10E23\libraries\SparkFun_9DoF_IMU_Breakout___ICM_20948___Arduino_Library-1.2.7\src/util/ICM_20948_DMP.h:644:17:
note: in expansion of macro 'Z'
  644 |         int32_t Z;
      |                 ^
exit status 1
Error compiling for board MSP-EXP430FR5994LP.

Thanks & Regards
Tarun Karak

StefanSch, 2021-10-06:
Looks like that the ICM 20948 libraries uses some definitions which are also used by the MSP430 device header file.

If you add this two lines before the ICM_20948.h include 
#undef ID
#undef Z
#include "ICM_20948.h" // Click here to get the library: http://librarymanager/All#SparkFun_ICM_20948_IMU

it should work.
By the way what was required to get the devices.csv error solved.

tarunrtes, 2021-11-05:
> Dear sir,
> I can send "Hello World " through UART to the serial monitor. But I am not
> getting ICM 20948 sensor data in the serial monitor.
> I have added these two lines before the ICM_20948.h include as you
> suggested.
> #undef ID
> #undef Z
> And replace devices.csv inside this path with the one you have attached.
>
> 'KARAK/AppData/Local/Energia15/packages/energia/tools/msp430-elf-gcc/
> 9.2.0.50/msp430-gcc-support-files/include/devices.csv'
> https://github.com/energia/msp430-lg-core/files/7291406/devices.csv (now
> using this devices.csv file)
> I am attaching the previous devices.csv.
>
> Thanks & Regards
>
> Tarun Karak

Dear sir,
I can send "Hello World " through UART to the serial monitor. But I am not
getting ICM 20948 sensor data in the serial monitor.
I have added these two lines before the ICM_20948.h include as you
suggested.
#undef ID
#undef Z

Thanks & Regards

Tarun Karak